nand2tetris/web-ide #18: VM Simulator

DavidSouther, 2022-05-30:


Goblinlordx, 2023-03-24:
Hi~ I just wanted to put it out there that I have a VM implementation for this that I think could potentially be useful. It was pretty... minimal to say the least and it has been a few years since I did the implementation (prior to me using TS much).

Either way, you can find it here:
https://github.com/Goblinlordx/hvm/blob/master/packages/vm/

Really, the core is here:
https://github.com/Goblinlordx/hvm/blob/master/packages/vm/VM.js

Goblinlordx, 2023-03-24:
Hmmm~ I will need to check the book references. It was intended to simulate all the hardware and run actual code with video output and inputs. The "video" output just copies the framebuffer data and renders to a 2d canvas context. There is also functionality for dumping memory and registers for introspection and/or live view of those when running. Though I never did get around to implementing a "real" usable UI for it.

Goblinlordx, 2023-03-24:
Ah~ I understand (looking back at the book). Sorry~ My misunderstanding.

closed 2024-04-10 by DavidSouther