nand2tetris/web-ide #58: Error messages / warnings

Shimon-Schocken, 2022-07-09:
We will have quite a few error messages and warnings.

I suggest that we pool all these messages in one file that I can easily edit. 

For each message, we have to record at least three things (in this file):

The message context (for us)
The message text (what the user sees)
Message ID (unique integer)

David -- please make this file and send me a link.

DavidSouther, 2022-07-09:
You're looking for an translations file, to keep user-visible strings consolidated for review?

DavidSouther, 2022-07-10:
I've started a branch with resource bundles in https://github.com/DavidSouther/computron5k/pull/61

The tagged user-visible strings are in https://github.com/DavidSouther/computron5k/blob/bb8985ce3feb817b130e2cfd598a0c4a863114d6/src/locales/en/messages.po

DavidSouther, 2022-08-02:
Additionally, there are now tests for compiler errors in https://github.com/DavidSouther/computron5k/blob/main/src/languages/hdl.test.ts#L36-L56

closed 2024-05-07 by DavidSouther