nand2tetris/web-ide #69: File system

Shimon-Schocken, 2022-07-11:
By "file system" I refer to the means that the IDE provides for organizing, creating, and modifying .hdl, .tst, .cmp, and .out files (projects 1, 2, 3, 5), and .hack and .asm files (project 5).  Here is a short list of desired services:

1. Users should be able to create, edit, and delete files.
2. Users will normally get default stub files that they will then extend.
3. (Nice to have) Users should be able to edit and manage their files outside the IDE.
4. (Longer run) Users should be able to use the IDE on mobile devices.

I believe that presently, all file access is done though the IDE. This will suffice for 90% of the users, but what about users who want to create their own chips? For example, build a different computer, or an extended ALU, or whatever? 

And what about users who want to create their own .tst files, or perhaps test the same chip using more than the single given .tst file?

Another issue that concerns me is the files location. David -- please give us some details. Are the files stored in a cloud-based server? Does this mean that in the long run we will be responsible (operationally and legally) for storing all the files of all the IDE users? Clearly, it will be better to let each user worry about his/her files.

Perhaps the best approach is to allow the user to enter a workspace path to the IDE, and have the IDE access all the files in this path, using a files explorer GUI and "loading / saving / etc." features. Can a JS-based system get such access to the host file system?  Or to the user's cloud-based file system?

It will help to start by making some educated guesses about how students will access their files in the next ten years.

I suggest that we start a conversation about this. Thx -- Shimon

DavidSouther, 2022-07-12:
In this version, the browser's [local storage](https://developer.mozilla.org/en-US/docs/Web/API/Window/localStorage) has a simulated file system. This means that the program can talk about "/projects/01/Not/Not.hdl", but that content is saved in the browser's local storage. This is a small (<5MB) location that is _per browser_ (not shared over the internet, not tied to any user accounts, completely local to the computer) area for storage. This works for both desktop and mobile (and there is responsive design for mobile views already).

Once you have selected a chip from the project and chip dropdown, pressing the "Save" button will overwrite the hdl, cmp, and tst files from the textareas. Adding a button to create a New chip is something we could do, as well as loading that chip when the builtin isn't otherwise available. Multiple test files for a single chip would take more design and work.

There are no hooks to allow a website to ask for an entire section of a user's file system, but we can ask for a dialog to read from a single directory or file. So we could do something like "Import from local file system", to initialize chips, programs, or memory. But building an entire hosted filesystem is beyond what I am comfortable supporting for this project.

At this point, for specific issues, I think we can add these?

1. Create and Delete custom chips, using the existing local storage solution.
1. Fall back to loading custom chips when Builtins aren't available.
1. Create and Delete multiple tst/cmp combos for a single chip, using local storage.
1. Import files from host file system
1. Export files for download to host file system

DavidSouther, 2023-01-02:
Closing as we've discussed thoroughly and are moving to the VSCode extension for "file system" management

closed 2023-01-02 by DavidSouther