nand2tetris/web-ide #70: Built-in chips with GUI effects: Implementation note

Shimon-Schocken, 2022-07-12:
The built-in chips that have GUI side effects are: 

- ALU
- ARegister
- DRegister
- PC
- RAM8
- ROM32K
- Screen
- Keyboard

The GUI side effects are displayed when the user opens one of these chips, or when an HDL program invokes it. 

I suggest that we implement the GUI side effects by displaying graphical objects that hover over the IDE and can be moved by the user around the screen.  If more than one chip is opened/invoked simultaneously, we can display a canvass and arrange the graphical objects inside this canvass, in some opportunistic layout (which depends on which chips are presently in use).

Shimon-Schocken, 2022-07-13:
ARegister, DRegister, PC, RAM8, and ROM32K: **Agreed**.

ALU has the ALU-like chip rendering, which we can replicate, but we might also want to play with it in other forms. **Agreed, let's experiment with other ALU images / renderings**. 

Screen: **Rendering it on the top right above Test is the natural place**. I assume that it will force pushing the Test and the other areas below it downward, and use scrolling? The problem with this is that the IDE layout will become unfamiliar and "unstable". I realize that drag and drop interfaces are less desired, but maybe the Screen and the Keyboard can be exceptions, since they are, literally speaking, peripheral I/O devices (unless we find a better solution). 

Keyboard: **Indeed in the desktop IDE implementation the Keyboard sucks. Your proposed rendering is much better.**

DavidSouther, 2022-08-08:
See #99, #100, and #101 for specific follow ups.

PR #102 (2022-08-08): Feat/visualizations
Visualizations panel for all chips. Closes #70, with follow-up work in others.
closed 2022-08-08 by DavidSouther