nand2tetris/web-ide #78: Grading

Shimon-Schocken, 2022-07-18:
Sorry that I have to raise this issue, but, here goes. Many schools use Nand to Tetris in the context of academic courses. Students are expected to submit their HDL programs for grading (typically, wrapped in a zip file like project1.zip). The grading can be done automatically, by executing behind-the-scene test scripts (presumably, more elaborate tests than the supplied scripts), and manually, to review the code quality.

I suggest that we use the opportunity of building this new IDE to start thinking afresh and creatively about grading practices. It will be wonderful if we could supply teachers with a tool that manages the automatic grading process. At minimum, we can think what it will take to at least interface with such a tool, and make sure that we have this interface in place.

DavidSouther, 2022-07-18:
I don't see any need to apologize, this is a great issue enhancement! I think the framework we have for testing the parts of the tool lends itself to reworking into an automated grading tool. Whether that's as a stand-alone command line program, a tab on the online IDE, or a stand-alone cloud service.

DavidSouther, 2022-11-15:
First pass at a CLI grader (with Java ide shadow mode) is in! #167

DavidSouther, 2023-01-07:
Closing; please open new issues for bugs or feature enhancements

closed 2023-01-07 by DavidSouther