nand2tetris/web-ide #79: Where to "store" the built-in chips

Shimon-Schocken, 2022-07-19:
I suggest putting all the built-in chips in a read-only "folder" which is at the same level as that of "Project 1", "Project 2", ...
The folder name can be "Built-in chips".

So, if you want to load, and play with, a built-in chip, say "RAM8", you invoke the top-left menu item, select the "Built-in chips" folder (which can be the first or last in the list), and then from it you select the RAM8 built-in chip.

There should be an option (advanced) to add new built-in chips to this folder, e.g. when one wants to build a computer other than Hack, or when one wants to add a new peripheral device to Hack, but we can worry about this later.

DavidSouther, 2022-07-19:
Let's hold off adding this until we have a more complete conversation on #69 creating a unified file system. The current project & chip selector is hard-coded, so anything here would be busy work without that design in place.

Shimon-Schocken, 2022-07-19:
Sure

DavidSouther, 2022-08-05:
The built in chips are already "stored" in memory, and available by changing the body from `PARTS: ...` to `BUILTIN;`. Rather than going through our fake file explorer, perhaps making that approach more obvious or documented?

Shimon-Schocken, 2022-08-11:
Please see what I wrote at the beginning of this thread. From an educational standpoint (for students and instructors alike), it's important to be able to access, and experiment with, built-in chips directly and explicitly, without messing up with their HDL. 

Therefore, I request that we implement what I described above. I realize that "storage" is hard wired and that the "directory structure" is contrived, that's OK for now.

Later on, and perhaps not by this team, we'll worry about allowing users to define and add built-in chips not mentioned in the book.

DavidSouther, 2022-08-15:
In #105, I added a button to toggle between using the user-defined and the builtin chips. When toggled to builtin, it disables & grays out the HDL box, keeps the input pins, and allows the user to click around & run tests against the builtins. Then going back to the HDL version, lets them continue editing their HDL.

DavidSouther, 2023-01-02:
Closing this as we've got the grader now, and starting the extension.

closed 2023-01-02 by DavidSouther