nand2tetris/web-ide #126: EVAL button / Pins behaviour

Shimon-Schocken, 2022-08-23:
The EVAL button should move from the HDL pane to the CHIP pane.

When you load a new chip which is not builtin, not implemented, or has an implementation (HDL) error, the EVAL button should be disabled (dimmed). If you try to click it, you should get the message "Incomplete / buggy HDL implementation; Cannot evaluate the chip logic." 

Presently, when you change the value of some input pin, the output and internal pins respond instantaneously, making the EVAL button unnecessary. The behaviour should be asynchronous: The output and internal pins must respond to input changes only when EVAL is clicked. I realize that this behavior is "less cool", but it gives the user a better sense of control. Also, it is pedagogically preferable. Here is how I suggest handling it:

Let's use the terms "active" and "dimmed" to describe the colors of the pin values. Presently, these colors signal that the former can be changed and the latter is "read only". I like this distinction, but we'll have to give it away for signaling consistency, as follows. The input pins should always be active. The internal and output pins should be dimmed by default. When EVAL is clicked, the internal and output pins respond to the input changes, and become active. They remain active until the user changes one or more bits in one of the input pins. At this point the internal and output pins become dimmed. They remain dimmed until the user clicks EVAL.

This implies that when the chip is in a consistent state (after EVAL has been clicked), all the pins (input, internal, output) look the same in terms of background and foreground colors. This is fine.

DavidSouther, 2022-08-25:
Should we move the clock buttons with the eval button, or into the Input Pins pane with special rendering for CLOCKED chips?

closed 2022-08-26 by DavidSouther

Shimon-Schocken, 2022-08-26:
Very nice fix... Thx!

PR #129 (2022-08-26): Feat/chips/eval button
* Chips compile on edit
* Chip panel shows "Invalid chip" when parse or semantics are incorrect
* Chip panel tracks "pending" changes to in pins, and only evals when clicking the eval button
* Eval button moved to chip panel header
* Clock buttons moved into Pins table 

Closes #126