nand2tetris/web-ide #127: Comparison / Diff

Shimon-Schocken, 2022-08-23:
I wish to give the user a better sense of control, and gain some more backward compatibility with the desktop IDE.

Please rename "Comparison" to "Compare file", and "Diff" to "Output file". I realize that these are not real files, but maybe they will be in future versions of the IDE.

When the test script runs, create the output file incrementally, line by line. When there is a diff, highlight the last line in the output file (where the diff is), and stop the test script execution with the message: "Output line is not the same as compare line; Test script execution halted".

The output file should look exactly the same as the compare file, including the variable header line and the "|" separators. To make this point clearer: If you could have saved the output file of a well-define chip, you could have renamed it and used it as a compare file (in fact, that's how we created the compare files in the desktop IDE).

DavidSouther, 2022-08-24:
I started with that comparison experience with this diff table, but wanting to move away from the "File" metaphor. When I first was going through the book, I actually found it really frustrating that the test would stop on my first error, rather than showing me all my errors. I also thought that it being a separate file was a little silly, for the common case of iterating on my HDL. Those guided me in this version to wanting to have all the errors, and see the entire panel of "good" and "bad" at once. Indeed, when I was building out the CPU, at one point I had all the ALU & instruction logic wired up, but I hadn't yet wired the PC to its output. Seeing the entire PC column go from "Red" to "Green" when I did that last connection was really cool & a good feeling.

So, I understand the history you're bringing from the desktop IDE, but I want to push back a little bit on the diff table and ask you to use it a bit before we insist on changing it back. (And, indeed, under the hood it's implemented as two text buffers, & I did have the display of literal files at one point!)

On the flip side, opening something like Memory and seeing "218 failures" before you even run the chip is probably daunting. At least, that shouldn't show up until the test has been run at least once. Maybe there should be a "run all" / "run until first error" toggle?

For the text changes, those are done in https://github.com/DavidSouther/computron5k/commit/7f86cf43114c42689389f2dfc0ec5f21ebf99637#diff-e460ab2117f8fb32489728a1be4b98389532478176c48221c988783053535bcd

Shimon-Schocken, 2022-10-11:
I am totally aware that I am biased toward some of the antics of the
desktop version. And, in general, I am still somewhat bewildered by "the
web".

I love working with text files for their modularity, simplicity, and
scalability. There will always be users who may want to (i) write their own
test scripts, (ii) add new chips (HDL and/or builtin), e.g. an interrupts
register, and (iii) build some chips that have nothing to do with Hack.
Admittedly, the vast majority of users will not feel that they are missing
anything with the new IDE, and that's the user base that we must worry
about. The loss of generality will be more than compensated for by the gain
in usability. But, I did not lose hope yet for entertaining the power users
as well. Maybe your suggested VSCode plugin approach is the solution. I
like VSCode very much.

The current DIFF solution has two problems, in my opinion (the ominous
error message at the outset is a minor issue). First, it makes no sense to
display outputs before the simulation (and the "output" commands) start
executing. Second, the color coding and the need to scroll horizontally are
cryptic. I suggest the following design:

Don't show any output until the script starts executing. Have the script
build the output file incrementally, line by line, each time an "output"
command is encountered. Complete the simulation and print the entire file,
including all the errors. When there is an error, color the incorrect
output line in red, and the corresponding compare line in blue (each in its
own pane). Alternatively, color only the DIFF bits (not the entire lines).
I believe that scrolling through the compare and output files vertically
and observing these pairs of diif lines will be a most intuitive way to
discern errors and understand the testing logic.

Builtin chips: Still can't see them! When I set RAM8 to builtin, the
Visualization button has "None" under it. When I click the Visualization
button, the "None" text is gone, and nothing is displayed. See below.

-- Shimon

[image: image.png]

DavidSouther, 2022-10-14:
For the test output:  https://github.com/DavidSouther/computron5k/pull/141

For the visualizations: It looks like I did something to all the BULTIN visualizations. Chips with PARTS have visualizations for their BUILTIN parts, but chips using BUILTIN don't have visualizations. I opened https://github.com/DavidSouther/computron5k/issues/142

PR #141 (2022-10-16): Diff table feedback
Only show errors for completed out lines. 
Show errors stacked in a single cell.

Closes #127 

A passing diff table

[image]

A diff table with failures

[image]

For scrolling the table, I think this is the best we're going to get from the web layout. Having the flexible panels in VSCode will be a plus.
closed 2022-10-16 by DavidSouther