nand2tetris/web-ide #151: Input / output / internal pin formats

Shimon-Schocken, 2022-10-23:
It seems to me that presently the format of all the pins is binary by default, with no alternatives.

It suggest binary by default, with an option to switch between binary and decimal. In the book / projects we rarely mention Hexa, and I don't want to add unnecessary clutter.

The decimal option is quite handy when teaching / demonstrating certain chips.  

UI: I suggest putting a button with the dimmed text "B/D" on the right of the labels "Input pin/s", "Output pin/s", "Internal pin/s".

DavidSouther, 2022-11-02:
This behavior is available from the Memory builtin visualization, as well as the "Util" page (which I can add to the VSCode extension). When in Hex / Dec modes, what interactions do you think are appropriate for editing? (Recall I had the +1 button for incrementing the binary format; we agreed it didn't work great, but what would we do instead?)

Shimon-Schocken, 2022-11-07:
Decimal comes handy when a teached demonstrate several things in class (or students experiment on their own). For example, when you test the ALU, the x and y inputs and the ALU output are easier to read in decimal. Likewise, when you enter values into a register / RAM device, it's sometimes easier to explain and follow when both the values and the address are specified in decimal.

DavidSouther, 2022-11-07:
Yes, having the different formats seems very useful. I'm asking, what user operations should be available when in these other modes?

Shimon-Schocken, 2022-11-07:
Good question... When the binary input has length 1, the only valid decimal values are 0 or 1.

I am not sure yet what to do when the length is n > 1. If it's an address field, valid values are between 0 and 2^n - 1. In all other cases it should be between -2^(n-1) to 2^(n-1) - 1. But I don't know how to detect that an input is supposed to be an address or not, that's a tricky issue. Let's discuss when we meet.

Shimon-Schocken, 2022-11-07:
I suggest that we solve this problem by cheating, as follows:

If the chip is a Nand to Tetris chip, we can flag (internally) each chip input pin as "value" or "address", and apply the validation checks accordingly (in decimal mode).

If the chip does not belong to the Nand to Tetris chip set, we'll disable the binary/decimal choice, and use only binary as (the only) default.

Not a very elegant solution, but it will be effective 99% of the time, since users rarely build chips of their own. And if thy will, they will still have all the functionality, except for this little B/D feature.

DavidSouther, 2024-05-07:
@netalondon has implemented this

closed 2024-05-07 by DavidSouther