nand2tetris/web-ide #154: Compare files / Output files

Shimon-Schocken, 2022-10-23:
When the simulation is successful, the compare and output files should be exactly the same: Line numbers, column dividers (vertical bar characters), blank lines, etc.

By the way, some scripts do have blank lines. Is this intentional? To make them more readable?

DavidSouther, 2022-11-01:
Is this not the case currently? The compare panel adds formatting and highlighting, but the underlying files (in memory) are otherwise identical. I did edit some scripts for readability when copying them from the course site to this repo.

Shimon-Schocken, 2022-11-07:
They are the same contents wise, but not format wise. Let's talk about it in our next zoom meeting.

DavidSouther, 2024-05-07:
This format was implemented at some point

closed 2024-05-07 by DavidSouther