nand2tetris/web-ide #187: Web IDE drops "Builtin" toggle switch during edge case

njkrichardson, 2023-01-31:
Hi there, 

The following sequence of steps causes the "Builtin" toggle switch to disappear unless I reload the page: 
1. Select some chip
2. Toggle builtin ON 
3. Select another chip (before turning builtin OFF) 

At this point, the builtin toggle disappears altogether and does not reappear until the page is reloaded.

DavidSouther, 2023-02-02:
Thanks for the bug report! A variation on this can also prevent changing the project selector.

Shimon-Schocken, 2023-02-03:
Following up on this issue: The students will start working on Project 3 next week. If it will be fixed by then, great. If not, we'll notify them about the issue.

DavidSouther, 2023-02-03:
This affects any project on the Chips page. I'll have time this afternoon to look into it.

PR #189 (2023-02-05): Always load the new chip, then switch to the builtin if necessary.
Closes #187.
commit 096dd79c (2023-02-05): Always load the new chip, then switch to the builtin if necessary. (#189)

Closes #187.
commit c42b83f4 (2023-02-06): Always load the new chip, then switch to the builtin if necessary. (#189)

Closes #187.
commit aa914e35 (2023-02-06): Always load the new chip, then switch to the builtin if necessary. (#189)

Closes #187.
closed 2023-02-05 by DavidSouther