nand2tetris/web-ide #188: CPU implementation that passes test on Java simulator but fails on Web IDE

RJWoodhead, 2023-02-02:
Ran this as a torture test and there seems to be a discrepancy about when registers are updated during the tick-tock cycle. No idea if the problem is with the original Java implementation, the new IDE, or my CPU implementation is just plain weird.
[CPU.zip](https://github.com/nand2tetris/web-ide/files/10571441/CPU.zip)

DavidSouther, 2023-02-03:
It is not your code per se, but rather the tests for the current desktop project version. Shimon, Noam, and I discussed this on august of last year. The test cases in the web IDE are correct, and you'll see that they update the DRegister on the tock (clock t+) step. The test files included in the book have the DRegister's output updating during the tick (clock t) step.

I have labeled it "bug" "book compatibility" though IMHO the solution is to change the files sent to the desktop version.

DavidSouther, 2024-06-03:
Closing this as the web IDE has been released as the recommended IDE going forward. It is still available for historical record.

closed 2024-06-03 by DavidSouther