nand2tetris/web-ide #207: [bug]: ctrl + c to copy crashes the instance and requires a reloading of the page

GuyOsovsky, 2023-12-25:
### Program

Hardware Simulator

### Interface

Website (https://nand2tetris.github.io/web-ide)

### Contact Details

guyosovsky@gmail.com

### What happened?

A bug happened!
when trying to ctrl + c to copy the hdl code it crashes the instance and requires to reload the page

### Do you want to try to fix this bug?

- [ ] I want to try to add this feature!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DavidSouther, 2023-12-25:
Guy, could you please share your OS and browser? Also please share a bit more about what you saw when it "crashe[d] the instance" - did the browser stop working, or were you unable to make progress using the hardware simulator?

DavidSouther, 2024-04-10:
I have been unable to reproduce this

1. Open Hardware Emulator
2. Load default "Project 5" -> "Computer"
3. Run ComputerAdd.tst
4. Select all text in hardware editor
5. Ctrl+C -> no effect on running program

If you can reproduce it, please re-open the bug with steps!

closed 2024-04-10 by DavidSouther