nand2tetris/web-ide #313: [bug]: expected end of input

isaaclyman, 2024-05-20:
### Program

Assembler

### Interface

Website (https://nand2tetris.github.io/web-ide)

### Contact Details

mail@isaaclyman.com

### What happened?

In order to solve the "Mult" project in chapter 4, I opened `projects/04/Mult/Mult.asm` in the Assembler and pasted in the following code:

```
@R2
M=0

(LOOP)
  @R1
  D=M
  @END
  D;JEQ
  
  @R0
  D=M
  @R2
  M=M+D
  @R1
  M=M-1
  @LOOP
  0;JMP
(END)
  @END
  0;JMP
```

Below the editor window, I immediately get the message:

> Line 22: expected end of input

The online Assembler will not translate this code. However, the legacy Assembler translates it with no issues at all.

### Do you want to try to fix this bug?

- [ ] I want to try to add this feature!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DavidSouther, 2024-05-20:
@isaaclyman thank you for raising the issue!

@netalondon can you take a look and see why we can't handle multiple empty lines in the assembler?

isaaclyman, 2024-05-20:
I've tried a couple of things, by the way:

- Removing the blank line before `@R0` doesn't change the error message.
- Typing in the code manually, instead of pasting it, doesn't seem to make any difference.

DavidSouther, 2024-05-20:
The error I see is actually on line 13:

[image]

Which can be "fixed" by changing that to M=D+A

[image]

So we need to update the parser to include all the permutations in https://github.com/nand2tetris/web-ide/blob/main/simulator/src/languages/grammars/asm.ohm#L24-L35 and also that error message was very poor.

netalondon, 2024-05-21:
@DavidSouther I think it's intentional that we allow only D+M as specified in the book.

I'll get working on having a more user-friendly error message for this.

isaaclyman, 2024-05-21:
I would say the error message itself isn't the only problem. Observe how the red underline is on the wrong line below:

[image]

If you actually count out the lines (which has to be done manually, since line numbers aren't shown), line 21 is where the problem is. But the red underline appears at the end of line 18, which confuses the situation further.

PR #323 (2024-05-24): Improve assembly error reporting
Closes #313
closed 2024-05-24 by DavidSouther