nand2tetris/web-ide #325: Download button does not work in online editor[bug]:

ananth521, 2024-05-25:
### Program

None

### Interface

Website (https://nand2tetris.github.io/web-ide)

### What happened?

The Downoad button in online editor does not work.

### Do you want to try to fix this bug?

- [ ] I want to try to add this feature!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DavidSouther, 2024-05-26:
In chrome, I am able to download the project files:

[image]

You may be prompted to allow downloads, for instance, in Safari

[image]

after which the download completes.

[image]

It looks similar in Firefox:

[image]

DavidSouther, 2024-05-26:
If you are able to reproduce this issue with steps or screenshots, please do re-open this ticket!

closed 2024-05-26 by DavidSouther