nand2tetris/web-ide #326: Stub file input and out pins in the head of chip definition don't match with chip requirement[bug]:

ananth521, 2024-05-25:
### Program

None

### Interface

Website (https://nand2tetris.github.io/web-ide)

### What happened?

Stub file input and out pins in the head of chip definition don't match with chip requirement.

### Do you want to try to fix this bug?

- [ ] I want to try to add this feature!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DavidSouther, 2024-05-25:
Could you help us out and let us know which chip and project you're looking at?

ananth521, 2024-05-25:
In project1, in online editor, the stub file `in` and `out` definitions does not match with chip requirement. I had to download project1 files from software suite and replace the stub file in online editor with those downloaded from software suite. Not a big issue but it is what i have observed. The stub file of project1 stays the same in online editor, whatever the Chip selection is, from the drop down list.

DavidSouther, 2024-05-26:
First, please reset the project files from the settings menu:

1.  [image]

2.  [image]

Then, please confirm which chips and pins, specifically, are you seeing a difference between the book and the web ide?

---

Note that some files did change between the desktop "Projects" package and the web ide, however, the changes should be purely cosmetic. For example, in the projects download for the desktop edition, the given And chip has this stub:

```
// This file is part of www.nand2tetris.org
// and the book "The Elements of Computing Systems"
// by Nisan and Schocken, MIT Press.
// File name: projects/01/And.hdl

/**
 * And gate: 
 * out = 1 if (a == 1 and b == 1)
 *       0 otherwise
 */

CHIP And {
    IN a, b;
    OUT out;

    PARTS:
    // Put your code here:
}
```

Which in the web IDE has been streamlined and updated to 

```
/**
 * And gate: out = 1 if {a==1 and b==1}, 0 otherwise
 * And gate: if {a==1 and b==1} then out = 1 else out = 0
 */

CHIP And {
    IN a, b;
    OUT out;

    PARTS:
}
```

Which aligns with Appendix 4 from the second edition of the book

[image]

closed 2024-07-12 by DavidSouther