nand2tetris/web-ide #349: [bug]: Can not open a folder in Jack Compiler

Arcadonauts, 2024-06-07:
### Tool

Jack Compiler

### Interface

Website (https://nand2tetris.github.io/web-ide)

### Contact Details

nickfegley@hotmail.com

### What happened?

I clicked the folder button in the Jack Compiler, expecting it to open the file dialog, but instead it did nothing. This is not an issue in the other views (CPU Emulator, Assembler, etc.) 

### Additional Comments

In the javascript console (Ctrl + Shift + i), the error is: `Uncaught (in promise) TypeError: window.showDirectoryPicker is not a function`

### Do you want to try to fix this bug?

- [ ] I want to try to add this feature!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DavidSouther, 2024-06-07:
Thank you for the bug report - could you confirm which browser and version you're using?

Arcadonauts, 2024-06-07:
For what it's worth, I just tested it on Chrome on my machine and had no issues.

Arcadonauts, 2024-06-10:
I know this isn't the proper place to ask this question, but I'm not sure what _is_ the proper place, so please excuse this violation:

As I've worked through the course, I've kept a private list of issues I've come across with the Web IDE. The list is about 20 items long, and I'd like to know the appropriate way to share it. Should I open 20 different issues? Should I open one big issue? Should I pick my top 3 (or 5 or 10) issues and just submit those?

The IDE is awesome, and I don't want to appear ungrateful for the hard work the team has put into it. What is the proper etiquette here?

netalondon, 2024-06-10:
@Arcadonauts no worries, thanks for reaching out :)

The preferred way would be to open a separate issue for each item in your list.
No reason not to submit all of them.

DavidSouther, 2024-06-10:
@Arcadonauts I see you starting to put in issues - thank you! We'll go through them, and we'll either work on a fix or let you know why we want the feature to be that way, and find another way to address your concern.

DavidSouther, 2024-06-10:
@Arcadonauts thank you for your kind words, and for your excellent bug reports. As a small project, the only way we find bugs we haven't seen is from community users like you who've taken it in different places than we expected! We'll get to work on these, and we might have more questions. If you have any interest in also doing some Typescript development, we'd love to see pull requests if you want to take a stab at fixing any of these yourself!

DavidSouther, 2024-06-10:
(whoops wrong button, only meant to comment!)

Arcadonauts, 2024-06-11:
@DavidSouther  "If you have any interest in also doing some Typescript development, we'd love to see pull requests if you want to take a stab at fixing any of these yourself!"

I have only a passing familiarity with Typescript, and all the low hanging fruit (e.g. https://github.com/nand2tetris/web-ide/issues/362) seems to be handled already by someone that almost certainly has more knowledge than I do, which is probably for the best.

I will, however, keep an eye out for any changes that I might be able to make.

DavidSouther, 2024-11-30:
This should be better supported with the latest Local File System settings options. Please open a new feature request with suggestions to improve that feature.

PR #419 (2024-09-10): Local FS UI 
Depends on #411 

Closes #349 
Closes #445
closed 2024-11-30 by DavidSouther