nand2tetris/web-ide #352: [feature]: Ability to add user-created helper chips

CodeQuiver, 2024-06-08:
### Tool

Hardware Simulator

### Interface

Website (https://nand2tetris.github.io/web-ide)

### What feature are you proposing?

I would love it if we had the ability to add our own chips to use in addition to the pre-defined project chips. This is a feature that exists in the legacy desktop application, but I couldn't find a way to do this in the web version. (not entirely sure about the book for "book parity", but the Coursera lectures reference the ability to do this).

For example, Xor16 is not implemented in Project 1 or 2, but I wanted to add it so I could use it in the ALU chip. 

This could be implemented by having a "make new" button to create a generic blank chip file in whatever project is already selected. Once named, it could be stored and downloaded the same as the rest of the files.

### Do you want to try to add this feature?

- [ ] I want to try to add this feature!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DavidSouther, 2024-06-25:
From @martybillingsley:

> What feature are you proposing?

The hardware simulator is currently limited to the files from projects 1-3 and 5. In order to implement and test a chip that is not part of those projects, you have to go to some pains to edit and overwrite the project files.

I have my students implement additional chips, both as assignments and on exams. It's more difficult in the new software.

Would it be possible to allow upload of local .hdl, .tst, and .cmp files within in the hardware simulator?

havardox, 2026-04-19:
I would really like the ability to create custom chips. For example, the ALU project, since there is no 16WayOr. Even the ability for the editor to read user-created HDL files when the project is stored in a local directory would be a great improvement.

PR #411 (2024-09-10): Add local file system functionality to all tools
Closes #352