nand2tetris/web-ide #356: [feature]: Save UI settings when switching between modes

Arcadonauts, 2024-06-10:
### Tool

None

### Interface

Website (https://nand2tetris.github.io/web-ide)

### What feature are you proposing?

When switching between modes (e.g. from the assembler to the CPU Emulator), all settings are lost. I'm worried this is unclear, so here's an example:

When writing assembly code, my work flow was something along the lines of:

- Assembler:
  * Set speed to fast
  * Translate All
  * Load to CPU Emulator
- CPU Emulator
  * Set speed to fast
  * Change view from "bin" to "asm"
  * Oops! I made a mistake, got to go back to...
-  Assembler
   * Fix my mistake
   * **Set speed to fast 
   * Load to CPU Emulator
- CPU Emulator
  * ** Set speed to fast 
  * **Change view from "bin" to "asm"

Actions marked with **asterisks could be skipped if the Web IDE remembered that I wanted the CPU Emulator to be fast and in asm.

The above is just an example, every tool "forgets" its settings as soon as you switch away. I lost a lot of work in the Bitmap Editor and switching between the Jack Compiler and the CPU Emulator when writing Jack code is a similar process.

### Additional Comments

This seems like a minor QoL request, but as I am not an especially good assembly programmer, I probably switched back and forth between these tools hundreds of times while hunting down errors, and having to make those same UI changes every time was very unpleasant.

### Do you want to try to add this feature?

- [ ] I want to try to add this feature!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

netalondon, 2024-06-16:
@DavidSouther how do you propose we handle state preservation of the bitmap editor?
Is there a better way than rewriting the entire page in React?

DavidSouther, 2024-06-17:
Looks like it wouldn't be too bad to store grid to localstorage in DisplayGrid, and read it in InitGrid?

DavidSouther, 2024-06-17:
"store" would be... well, it could be `JSON.stringify` the 2d-array of bools! At some point, we could bring it up to date with React and our `Screen` implementation.

PR #384 (2024-06-23): Save bitmap editor state in local storage
Closes #356
closed 2024-06-23 by DavidSouther