nand2tetris/web-ide #391: [bug]: Hardware simulator page becomes unresponsive

amcooper181, 2024-07-07:
### Tool

Hardware Simulator

### Interface

Website (https://nand2tetris.github.io/web-ide)

### Contact Details

amcooper181@gmail.com

### What happened?

Twice, the page has become unresponsive when working. The first time it occurred was when working on the Mux4Way16 in Project 1, and the second was when working on Bit in Project 3.

On both occasions, the website became slow to respond then stopped responding altogether. This occurred when adding code to the "Parts" section, not while running anything. Upon reloading, the page would not load and Google Chrome gave a "Page Unresponsive" warning (see image). 

[image]

When using another browser, the issue was not present (but all progress was lost). Restarting the browser, as well as the computer, made no change. The only way to rectify the issue was to clear cookies for the IDE webpage, losing all progress. 

The issue occurred on:

-  Google Chrome version 126.0.6478.127
- Running on an Apple Macbook Pro with the M1 Pro processor running macOS Sonoma 14.0

### Do you want to try to fix this bug?

- [ ] I want to try to add this feature!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DavidSouther, 2024-07-08:
From #392, "I tried clearing the chrome cache, and now the IDE runs perfectly fine on chrome"
From #391, "The only way to rectify the issue was to clear cookies"
From #357, "refreshing the page usually fixes the problem."

So there are differing reports of what will or will not fix the issue, it does seem to be generally related to storage access.

IvanKusturic, 2024-07-10:
[image]

DavidSouther, 2024-07-10:
@IvanKusturic that looks unrelated to the page unresponsive issue, could you first reload the page and if the issue persists, open a new bug?

SeppahBaws, 2024-07-10:
Hi, I stumbled upon this myself as well, and I think it's because of an infinite loop that is triggered when you have a component have an output connect to its own input.

If I put this in the `Not` exercise in `Project1`, the page hangs.
```
CHIP Not {
  IN in;
  OUT out;
  PARTS:
  Nand(a=in, b=myNand, out=myNand);
}
```

A way to fix it without having to nuke the browser cache and losing all the process is this:
- open a new tab
- open dev tools
- go to the IDE url (the page should now be unresponsive)
- go to the "sources" tab in the devtools, and press the pause button in the right column
- in the "application" tab, go to "local storage" and under the site data, find the chip you were working on
- edit the value of that entry so that it doesn't have a circular reference anymore.
- reload the page. the HDL should now have been updated and the page works as intended again 👍

netalondon, 2024-07-11:
Thanks @SeppahBaws! That was some very useful debugging :)

PR #401 (2024-07-11): Check for looping wires when building chip
Fixes #391
closed 2024-07-11 by DavidSouther