nand2tetris/web-ide #392: [bug]:

leon-biner, 2024-07-07:
### Tool

Hardware Simulator

### Interface

Website (https://nand2tetris.github.io/web-ide)

### Contact Details

leon.biner@gmail.com

### What happened?

The web IDE frequently freezes, and at some point just doesn't work. On Firefox, it froze when I was implementing the incrementor chip, and now the page doesn't even load, so I switched to chrome. I just finished project 3 and I went on to test the program counter and it now takes forever to load the page, and it freezes when I do pretty much anything. I tried to run it in Safari and it runs perfectly, so I thought there was an issue with how the website stores data. I tried clearing the chrome cache, and now the IDE runs perfectly fine on chrome, so I think there's something wrong with the data storing. Thank you.

### Additional Comments

I am not a backend web developper so I don't know how to fix the bug or if I even gave you guys enough information to fix it, but feel free to contact me if you need more details on my end.

### Do you want to try to fix this bug?

- [ ] I want to try to add this feature!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

netalondon, 2024-07-08:
Hi @leon-biner 
Thanks for the bug report!

I'm closing this issue as it looks like a duplicate of #391, but if you think otherwise feel free to reopen.

closed 2024-07-08 by netalondon

leon-biner, 2024-07-08:
Hi,

Thank you for the response. I was wondering if I would get notified if/when
this bug is fixed.

Thank you,

Leon Biner

DavidSouther, 2024-07-08:
Yes, if you go to issue #391, down the right side of the page, click "Subscribe" under notifications:

[image]