nand2tetris/web-ide #405: [feature]: A way to save user-input so one can leave and come back to their project, or do all the projects over a period of weeks.

GameDiscovery, 2024-07-13:
### Tool

Hardware Simulator

### Interface

Website (https://nand2tetris.github.io/web-ide)

### What feature are you proposing?

A way to load my .hdl files back onto the website. I can download .hdl files of my chips from project 1 and 2, but the website reset and lost all my input, and I cannot find a way to load the saved .hdl files back onto the website. I want the satisfaction of knowing I could run hack from all my user-created chips, and am considering switching to the desktop version just so I can save my files and know I'm using them.

Also, my incomplete work on the ALU was lost overnight. Something reset and there was no save feature to load my work from.

### Additional Comments

Thank you for writing the book and creating this software! I am learning so much by reading through and following the steps. I am currently almost finished on project 2, and got stuck trying to figure out the logic for the ng output. It was when I took a break for the night that my work was lost.

### Do you want to try to add this feature?

- [ ] I want to try to add this feature!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

GameDiscovery, 2024-07-13:
Since writing this, I've discovered where you've written that my data is stored in my browser memory, and I managed to find my project work. I was confused. I still wouldn't mind this feature for a little security, but I did not actually lose my work. I'll be careful to not clear my browser cache while I take this course.

DavidSouther, 2024-08-14:
@GameDiscovery apologies for the delay in responding, but I'm glad to see you've not lost your work. When we initially wrote this Web IDE, the only long term storage we had readily available was in the browser's local storage itself. However, improvements in web browsers have opened some doors for us, and we're actively working on adding a feature to allow you to open a folder on your local desktop to store your NAND2Tetris files - and even create and edit them outside the web IDE!

closed 2024-08-14 by DavidSouther