nand2tetris/web-ide #406: [bug]: CPU Emulator Freezing on Computer Test

zaidbahssali, 2024-07-17:
### Tool

Hardware Simulator

### Interface

Website (https://nand2tetris.github.io/web-ide)

### Contact Details

mrzaidbahssali@gmail.com

### What happened?

Unless I am missing something that should be more obvious, it seems I am experiencing a bug related to the application failing to load the Add.hack, Mult.hack & Rect.hack programs into the simulator for testing.

When trying to test the project 5 Computer chip in the Hardware Simulator, the test stalls while loading the binary program into the ROM. The browser does not freeze but nothing productive happens. The visual indicator on the test script simply remains on that line. This is the same regardless of whether I am using the built-in chip or my own. None of the test scripts are able to get past the line which loads the program into the ROM.

I am using Microsoft Edge, if that helps. The Hardware Simulator has functioned flawlessly in this browser since beginning the course.

### Do you want to try to fix this bug?

- [ ] I want to try to add this feature!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

netalondon, 2024-07-18:
Hi @zaidbahssali, thanks for the bug report!

It's possible that those specific files somehow got deleted from your browser storage.
Could you try resetting the files from the settings and confirm if that fixes the issue or not?

Note that this will delete all of your saved progress, so you'll have to back it up first. 
Sorry for the inconvenience.

Alternatively, you can manually add the files to local storage through the dev tools, if you'd like to avoid clearing your progress. If you're comfortable with doing that, let me know and I'll walk you through it.

DavidSouther, 2024-11-30:
This should be resolved in the latest release, after #450 was merged

closed 2024-11-30 by DavidSouther