nand2tetris/web-ide #414: [bug]: Text Cursor slightly offset on the Web IDE

willdoyle757, 2024-07-26:
### Tool

General

### Interface

Website (https://nand2tetris.github.io/web-ide)

### Contact Details

willdoyle757@gmail.com

### What happened?

For some reason the text cursor is slightly offset when I use the web IDE. Additionally, it doesn't match the correct position of the actual mouse. This seems to only happen of the web page as when I use any other website or tab the text cursor works fine. I tried refreshing the page and closing it but it doesn't seem to work.

### Do you want to try to fix this bug?

- [ ] I want to try to add this feature!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

netalondon, 2024-08-04:
Hi @willdoyle757 

Could you attach a screenshot or screen capture of this behavior so I can better understand what the issue is?
Is this just an issue with the appearance of the text cursor or does this affect your ability to edit text?

Thanks

de2Zotjes, 2024-08-12:
Just created a new bug which is likely the same issue.

Attached a screen capture there.

It makes editing quite hard, since new text does appear where you expect it to.

It only happens with the Monaco editor, so as a work around that can be disabled, but then automatic loading of new files on switching chip does not happen..

netalondon, 2024-08-14:
Thanks @de2Zotjes.

I'll go ahead and close this in favor of #424

closed 2024-08-14 by netalondon