nand2tetris/web-ide #415: [bug]:

Kushalpc, 2024-07-30:
### Tool

Hardware Simulator

### Interface

Website (https://nand2tetris.github.io/web-ide)

### What happened?

A bug happened!
  And(a= , b= , out= );f  
when I type something in out , the word goes directly out of the paranthesis

### Do you want to try to fix this bug?

- [ ] I want to try to add this feature!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

netalondon, 2024-08-04:
Hi @Kushalpc 

I'm not sure I understand what you mean.
Could you please attach a screenshot of this behavior?

Thanks

netalondon, 2024-08-14:
Closing this as a duplicate of #424

closed 2024-08-14 by netalondon