nand2tetris/web-ide #422: [bug]: Test script section of WebIDE is uneditable in Firefox on Linux

luc-lisi, 2024-08-06:
### Tool

None

### Interface

None

### What happened?

While running Firefox on Linux Mint, the test script editor is unavailable. Changing from restore -> edit does not change anything. Chrome does work and I can confirm that Firefox works when running on Windows. 

There are no warnings or errors in the browsers console.

[image]

System info: `
   Operating System: Linux Mint 21.2                 
    Kernel: Linux 5.15.0-91-generic
    Architecture: x86-64
`

Firefox version: 120.0.1 (64-bit)

### Do you want to try to fix this bug?

- [ ] I want to try to add this feature!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

netalondon, 2024-08-12:
Hi @luc-lisi, thanks for reporting this.

Could you try updating to the latest Firefox version and let me know if this issue persists?

I'm currently on Fedora and reverted to Firefox 119, where the test panel is indeed missing. However, it works fine with the latest version, 129.

luc-lisi, 2024-08-12:
@netalondon 

Ah yeah that was it. 🤦‍♂️ 

I'll go ahead and close this!

closed 2024-08-12 by luc-lisi