nand2tetris/web-ide #424: [bug]: cursor position displays incorrect when using Monaco editor

de2Zotjes, 2024-08-12:
### Tool

Hardware Simulator

### Interface

Website (https://nand2tetris.github.io/web-ide)

### Contact Details

de2zotjes@gmail.com

### What happened?

The cursor does not move enough when moving along a line. It moves 9 positions for 10 characters  entered (digits 1 through 0).
The only action in the screen capture below was entering the digits, did not move the cursor after entering the last digit.

[image]

### Additional Comments

Using windows10 (22H2) with chrome (Versie 127.0.6533.100 (Officiële build) (64-bits))

### Do you want to try to fix this bug?

- [ ] I want to try to add this feature!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

netalondon, 2024-08-14:
Hi @de2Zotjes,
I'm having trouble reproducing this consistently.

Could you please check if resetting the files through the settings solves this issue? This seems to have been the case for me.
Note that this will delete your progress so you'll have to back it up first.

Thanks :)

netalondon, 2024-08-14:
> Could you please check if resetting the files through the settings solves this issue? This seems to have been the case for me.

@de2Zotjes nevermind this isn't the case. I can reproduce it now.

netalondon, 2024-08-14:
@DavidSouther this seems to happen only for the "JetBrains Mono" font we're using, and works fine on any other font.
Do you have any objection to changing it?

DavidSouther, 2024-08-14:
I see a couple threads that might offer a solution

https://github.com/microsoft/monaco-editor/issues/1618

https://stackoverflow.com/questions/78534256/why-does-space-change-when-using-custom-font-in-monaco-editor-on-web

https://stackoverflow.com/questions/58271107/offset-between-text-and-cursor-with-the-monaco-editor-angular-under-chrome-m

Let's try doing this remeasureFonts call and see if that works?

netalondon, 2024-08-14:
@DavidSouther yeah that works, thanks!

PR #426 (2024-08-14): Remeasure editor font when needed
Fixes #424
closed 2024-08-14 by DavidSouther