nand2tetris/web-ide #425: [bug]:

happytomatoe, 2024-08-13:
### Tool

CPU Emulator

### Interface

Website (https://nand2tetris.github.io/web-ide)

### What happened?

When using firefox 129.0 (64-bit) on Win 11 when I go to https://nand2tetris.github.io/web-ide/chip and try to run a test I get the the following
[image]
When I try to open "Converter tool" I see an empty page. Console shows same/similar error. When using Linux fedora 40 or chrome everything LGTM. Is this expected? What browsers/platform does the web ide support?
    

### Do you want to try to fix this bug?

- [ ] I want to try to add this feature!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

happytomatoe, 2024-08-13:
Sorry, the tool should be Hardware simulator.

netalondon, 2024-08-19:
Hi @happytomatoe, Thanks for the bug report.

I apologize for the late reply, but I just got to testing this and can't replicate your issue.
Is this still happening?

happytomatoe, 2024-09-16:
@netalondon Let me try again. BTW Do you have plans to include support for firefox? I guess that would require a lot of effort. Currently I am writing a Jack game. I've tried to run it inside VM Emulator using firefox and it's lagging a lot compared to when using Chrome.

happytomatoe, 2024-09-16:
After short testing - 
I don't get error on this page anymore. I get this error on 
[image]
or when opening https://nand2tetris.github.io/web-ide/util

DavidSouther, 2024-09-20:
I am able to open and run the VM Emulator in Firefox. Please re-open if you can provide specific steps to reproduce the issue.

[image]

closed 2024-09-20 by DavidSouther