nand2tetris/web-ide #429: [bug]: 1-2 got 65535

ndNovaDev, 2024-08-16:
### Tool

CPU Emulator

### Interface

Website (https://nand2tetris.github.io/web-ide)

### What happened?

```
@1
D=A
@0
M=D
@2
D=A
@0
D=M-D
```
the expected value of D is -1, but got 65535

### Do you want to try to fix this bug?

- [ ] I want to try to add this feature!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ndNovaDev, 2024-08-16:
```
@1
D=-A
```
the expected value of D is -1, but also got 65535

netalondon, 2024-08-18:
Hi @ndNovaDev, 
`65535` is equivalent to `-1` using 2's complements, so this is actually the intended behavior.

If you feel there's more that needs to be said about this, feel free to reopen the issue.

closed 2024-08-18 by netalondon

ndNovaDev, 2024-08-19:
oh sorry, it's my fault