nand2tetris/web-ide #436: [bug]:Register chip section has the files for DFF and not Register

talasleman, 2024-08-25:
### Tool

Hardware Simulator

### Interface

Website (https://nand2tetris.github.io/web-ide)

### Contact Details

slemantala@gmai.com

### What happened?

In Project 3, of the hardware simulatior, sometimes when you change between files for a certain chip, the last implementation follows with it. For example, I was on DFF chip and changed to Register chip to implement it, but the DFF implementation showed up again (and the test files too) it is currently irreversible which means i can't implement the Register chip.

### Do you want to try to fix this bug?

- [X] I want to try to add this feature!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DavidSouther, 2024-09-16:
Thank you for opening this issue, @talasleman. I'm not able to recreate it from what you've written - can you include a screenshot of the issue if it persists?

DavidSouther, 2024-11-30:
Please feel free to include a screenshot or video and reopen this issue if it happens again.

closed 2024-11-30 by DavidSouther