nand2tetris/web-ide #437: [feature]:

benjaminclauss, 2024-08-25:
### Tool

CPU Emulator

### Interface

Website (https://nand2tetris.github.io/web-ide)

### Contact Details

benjaminclauss@protonmail.com

### What feature are you proposing?

Display comments in ROM.

### Do you want to try to add this feature?

- [ ] I want to try to add this feature!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maueroats, 2024-09-02:
Please clarify what this proposal means.

Give an example of what it would look like.

benjaminclauss, 2024-09-05:
from `Add.asm`

```
// Computes R0 = 2 + 3  (R0 refers to RAM[0])
@2
D=A
@3
D=D+A
@0
M=D
```

something like the following

[image]

This would allow users to understand what part of their code is running.

maueroats, 2024-09-06:
This feature request will require some actual work. Right now the "asm" view that you see is decompiled instead of being a view on the "original" source code. You can verify this by uploading a "binary" (decimal?) compiled file instead of an asm file. 

This proposal also needs to clarify what line numbers will be displayed on comment lines. 

I think a blank (no number) would be the least confusing.

I can think of a danger to displaying the original source code - two variables that are spelled similarly but assigned to different memory locations will be obviously different in the "decompiled asm" view but not the source view.

For example, the nonsense below: 
```
@GOOD_X
M=-1
@good_x
M=0
```

vs 

```
@17
M=-1
@18
M=0
```

I too would like the ability to show the original source, but I don't think it's a given that showing the source is a strict improvement over the current version.

benjaminclauss, 2024-09-10:
> Right now the "asm" view that you see is decompiled instead of being a view on the "original" source code.

That's unfortunate.

I think a big part of working on VM translator projects is the ability to step through compiled ASM files.

DavidSouther, 2024-09-16:
Thank you for the suggestion and discussion on its implementation! As @maueroats notes, the web IDE doesn't currently capture source code information by the time we get to the rendered view, but it is an interesting idea to follow. We've had several discussions on what making this system into a "more proper" IDE would be. Capturing the source map for variables & labels, as well as AST nodes for comments, would be a necessary part of that.

benjaminclauss, 2024-09-16:
sounds reasonable

Feel free to close.

happytomatoe, 2024-09-16:
@benjaminclauss  qq - have you already written VM translator? If yes how did you debug hack code inside CPU emulator?

happytomatoe, 2024-09-16:
Though I guess debugging is still debugging whether it's a vm translated code or just a handwritten assembly code

maueroats, 2024-09-16:
> @benjaminclauss qq - have you already written VM translator? If yes how did you debug hack code inside CPU emulator?

(Not a student.) I produced a version of the (my) source code with line numbers that correspond to ROM locations and follow that in parallel with the code in the IDE. Auto-generated comments for the VM instructions give good markers to follow at both the VM level and the Hack level.

benjaminclauss, 2024-09-17:
> @benjaminclauss qq - have you already written VM translator? If yes how did you debug hack code inside CPU emulator?

I finished recently.
https://github.com/benjaminclauss/nand2tetris/blob/master/virtualmachine/code_writer.go
It needs cleanup but tests are passing.

Debugging was a pain point, hence the creation of this issue.
Translating VM code yielded 100s of lines of ASM instructions which made it difficult to identify which logic was being run to validate its correctness **especially** with `JMP` commands.