nand2tetris/web-ide #441: [documentation]: Is there a table that compares desktop legacy version with web version?

happytomatoe, 2024-09-01:
### Tool

General

### Interface

None

### What feature are you proposing?

Hi! First of all thanks for this product. 
Is there a table that compares the features between desktop and web version? Why I am asking -  I've just finished writing vm translator and some portion of the time I've spent on debugging. I totally forgot that desktop version has breakpoints. That would make the process a little bit easier. I didn't see any mentions about breakpoints for web ide. Is this feature available in web version?

### Do you want to try to add this feature?

- [ ] I want to try to add this feature!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DavidSouther, 2024-09-16:
The current goal is "feature parity", so this would be a thing we've missed in the current IDE and should consider implementing or explicitly marking as unsupported. I'm going to update this issue to track breakpoints; if you notice other discrepancies, please let us know.

@netalondon @Shimon-Schocken please comment on where we should prioritize adding debugger breakpoints for VM and CPU Emulators?

happytomatoe, 2024-09-17:
@DavidSouther @netalondon How to pause a runner from inside vm.ts? I.E. in the case if the current running operation has a breakpoint

happytomatoe, 2024-09-17:
The same applies to CPU emulator

netalondon, 2024-09-17:
@happytomatoe If I understand you correctly, you can emulate this behavior by immediately returning from the `step` function here https://github.com/nand2tetris/web-ide/blob/main/simulator/src/vm/vm.ts#L634.
You can see this is what we do if the OS is blocked, for example when waiting for user input.

Should be pretty similar for the CPU.

Let me know if that answers your question.