nand2tetris/web-ide #443: [bug]: "set reset 1" unrecognied in CPU Emulator - Test Script

maueroats, 2024-09-02:
### Tool

CPU Emulator

### Interface

Website (https://nand2tetris.github.io/web-ide)

### Contact Details

I will follow this issue.

### What happened?

tl;dr - I think RAM should reset with `set reset 1`. This is shown in Figure A3.3 (2nd ed) but I could not find it mentioned in the text.

Behavior observed on: Chromum 128.0.6613.84 (Official Build) snap (64-bit). Ubuntu 22.04 system.

Observed this bug in the Web IDE most recently on September 1, 2024. The behavior is observed in Incognito mode (attempting to eliminate variables).

## Reproduce

1. Open Chromium in Incognito mode.
2. Go to `https://nand2tetris.github.io/web-ide/cpu/` 
3. ROM: Load some ROM file generated locally.
4. Test: Default -> Edit -> OK 
5. Insert `set reset 1;` as the first line, as shown in Figure A3.3 of the second edition.

Text area at the bottom of the screen shows:

      Failed to parse test - Line 2: expected ",", "!", or ";"

## Expected Behavior

Executing that line of the simulator sets all of the RAM values to 0.

### Additional Comments

This is useful when writing your own test scripts.

It wasn't clear to me when the memory was cleared in the test script execution. Not every rewind (which would be another logical choice). 

### Do you want to try to fix this bug?

- [ ] I want to try to add this feature!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DavidSouther, 2024-09-16:
@maueroats thank you for the bug report.

[image]

@Shimon-Schocken should we implement `set reset 1`, or auto-reset RAM to 0 when resetting tests?

netalondon, 2024-09-16:
Note that this is `ComputerMax.tst` - a test for the HDL implementation of the Computer chip. 
I think that `set reset 1` here refers specifically to the Computer input pin `reset`, and is not intended to be used in tests in the CPU emulator.

That said, @Shimon-Schocken suggested implementing a `resetRAM` command, which I'll do.

Additionally, the Java version doesn't reset the RAM when resetting the test.
Do we have any reason to change that?

maueroats, 2024-09-16:
> Additionally, the Java version doesn't reset the RAM when resetting the test.
Do we have any reason to change that?

Not resetting the RAM could cause the second run of a test to have different results than the first run, which would be very confusing to students. 

Unverified: I think you don't see this effect partially because some other part of the test process does reset the ram (perhaps the load?).

PR #462 (2024-09-17): Implement resetRam test command
Closes #443