nand2tetris/web-ide #446: [bug]: VM emulator skips a loop

happytomatoe, 2024-09-04:
### Tool

None

### Interface

None

### What happened?

[image]
Sample code
```
class Main {
  function void main() {
    while (true) {
      do Main.drawCloud(100);
      }
      return;
  }

  function void drawCloud(int location) {
    var int memAddress;
    let memAddress=16384 + location;
    do Memory.poke(memAddress+1, 448);
    do Memory.poke(memAddress+33, 1854);
    do Memory.poke(memAddress+65, 3075);
    do Memory.poke(memAddress+96, -32768);
    return;
  }
}
````
I am using Chromium Version 127.0.6533.99 (Official Build) Fedora Project (64-bit). I am also running web ide in incognito mode

### Do you want to try to fix this bug?

- [ ] I want to try to add this feature!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

happytomatoe, 2024-09-04:
nvm. I am running fibonacci sequence

happytomatoe, 2024-09-04:
qq - is it possible to run Jack files in web ide? If yes how?

netalondon, 2024-09-08:
You can compile your project using the Jack compiler and click the run button to load the compiled code to the VM emulator where you can run it.

closed 2024-09-08 by netalondon