nand2tetris/web-ide #447: [improvement]: Jack compiler is lagging with big files

happytomatoe, 2024-09-08:
### Tool

Jack Compiler

### Interface

Website (https://nand2tetris.github.io/web-ide)

### What feature are you proposing?

Hi! 
I am writing a game in Jack and currently I have 7 files with 7238 lines. 
```
~nand2tetris/jack/dino/src$ wc -l *
   574 Cactus.jack
   142 Cloud.jack
   707 Dino.jack
    25 Main.jack
    10 MathCustom.jack
    14 ScreenCustom.jack
  5766 Trail.jack
  7238 total
```

After every key stroke I need to wait about a considerable amount of time before I can continue.
[image]

Chrome shows this after I press run
[image]

How to recreate this?
I guess if you will create a lot of Jack code that will do or you can use the code from my repo. 

Why my files are so big? 
I've created an utility that converts images into jack code and some of the sprites are quite long (like trail width>2000 pixels) and I am drawing each pixel individually. 

qq - Is this request has low priority? Additional question - do you happen to know if Java tool is still maintained?

### Do you want to try to add this feature?

- [ ] I want to try to add this feature!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

happytomatoe, 2024-09-08:
Please also note that VMEmulator from Java tool fails because program counter gets into negative values. I forgot that a program cannot have more than 2^15 lines. Probably even less for vm code as it will be translated into assembly where this program counter also exists.

happytomatoe, 2024-09-09:
I guess the error messages for the jack compiler in the old version are better. Please feel free to close this issue.

DavidSouther, 2024-09-16:
@happytomatoe thank you for your issue report. 

We will certainly leave it open to track possible performance improvements for large Jack projects, and thank you for the link to your repo so that we have some to test against! 

PS The java tool is generally unmaintained and released "as is" in its current state - while the code exists in several places online, bringing it to "modern" standards was considered to be less important than this web-based interface.

closed 2026-10-21 by DavidSouther