nand2tetris/web-ide #454: [bug]: Sometimes when I run a jack program I get  Error: Current operation step beyond end of function operations (290 > 39)

happytomatoe, 2024-09-14:
### Tool

VM Emulator

### Interface

None

### What happened?

Sometimes when I run my jack program I get 
```

main.9984f23c.js:2 Uncaught Error: Current operation step beyond end of function operations (290 > 39)
    at get operation (main.9984f23c.js:2:468893)
    at x.derivedLine (main.9984f23c.js:2:475281)
    at S (main.9984f23c.js:2:238859)
    at Object.update (main.9984f23c.js:2:239816)
    at main.9984f23c.js:2:223865
    at produce (main.9984f23c.js:2:222483)
    at main.9984f23c.js:2:223845
    at Object.wa [as useReducer] (main.9984f23c.js:2:104836)
    at t.useReducer (main.9984f23c.js:2:203526)
    at q (main.9984f23c.js:2:223836)
```

What does error mean? 

### Do you want to try to fix this bug?

- [ ] I want to try to add this feature!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

happytomatoe, 2024-09-14:
Whole log
[nand2tetris.github.io-1726296150652.log](https://github.com/user-attachments/files/17001383/nand2tetris.github.io-1726296150652.log)

happytomatoe, 2024-09-15:
Somehow "this" pointer (address number 3 ) becomes 0 inside of IntQueue.clear
How to recreate this - compile and run [this](https://github.com/happytomatoe/nand2tetris/tree/982664a23b39fea875385f62c245ecf5786ed360/jack/dino-game/src)

DavidSouther, 2024-09-16:
@happytomatoe thank you for opening this issue. It will probably be easier to troubleshoot these if you're running in development mode locally. You can do this by installing Node.JS, cloning the repo, running `npm install`, and then `npm start` to have a local copy running. This will use source maps for the javascript, which include the original line numbers, and we can get a better look at the problem.

happytomatoe, 2024-09-16:
@DavidSouther Thanks for the suggestion. Only today I've realised how to debug web IDE using VS code. 
BTW I am no longer receiving this error. I don't know how it was fixed. 
Any comment on what [this error ](https://github.com/nand2tetris/web-ide/blame/fab8cec64af12cd36ae47b210ab2487f1c3930f4/simulator/src/vm/vm.ts#L455) means or situations that can lead to that?

closed 2025-08-10 by DavidSouther