nand2tetris/web-ide #463: [bug]: VMEmulator is running with duplicated labels

happytomatoe, 2024-09-16:
### Tool

None

### Interface

None

### What happened?

If we are compiling 2+ jack files with at least 1 label in each, in VM code we will have L0 label duplicated. Is this expected? Seem like it shouldn't run at all. 
Why am I asking? I am trying to use vm translator for compiled files and I get duplicated label exception. Probably I will need to provide a hotfix for this one. Probably prepending file name to a label, in some format, will suffice. 

### Do you want to try to fix this bug?

- [ ] I want to try to add this feature!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

netalondon, 2024-09-16:
Hi @happytomatoe, thank you for the bug report!

Just to make sure I'm getting this right, the duplicate label error is coming from the assemler, correct?

> Is this expected? Seem like it shouldn't run at all.

You're right, that's an oversight on our end. Both versions of the VM emulator actually know how to handle this case, so the issue went unnoticed, but it doesn't mean we should expect students' VM translators to do it as well.

So thanks for catching that!

netalondon, 2024-09-17:
@happytomatoe Oh yeah that makes sense as well. Thanks!

PR #465 (2024-09-17): Add class name as prefix to VM labels in the compiler
Fixes #463
closed 2024-09-17 by DavidSouther