nand2tetris/web-ide #470: [bug]: ChunkLoadError: Loading chunk 537 failed. (timeout: https://nand2tetris.github.io/web-ide/static/js/537.c02a5bfa.chunk.js)

talhol, 2024-09-17:
### Tool

Hardware Simulator

### Interface

Website (https://nand2tetris.github.io/web-ide)

### Contact Details

tal.hollander@mail.huji.ac.il

### What happened?

A bug happened!
the website is not working, I tries reseting the computer and it didn't help.

### Do you want to try to fix this bug?

- [ ] I want to try to add this feature!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

talhol, 2024-09-17:
> Hi @talhol <https://github.com/talhol>, thanks for the bug report.
>
> Unfortunately, I can't replicate this on my end,
> could be a caching issue?
>
> Do you have any unsaved work on there or could you try clearing the
> browser cache and see if that solves the issue?

Clearing the browser cache helped, but it deleted my work and the bug
showed up again after half an hour.

DavidSouther, 2024-09-18:
@netalondon it looks like we haven't trigged a release build in a couple months? I thought I had the github action figured out. I just triggered a build & deploy.

@talhol you shouldn't need to clear the entire browser cache, only skip caching when loading the page. To do this, while clicking the browser's refresh button hold either Ctrl (Windows + Linux) or Shift (Mac) keys. This should work for Chrome, Firefox, Edge, and Safari.

closed 2024-11-29 by DavidSouther