nand2tetris/web-ide #472: [bug]: Running VM emulator from main branch results in Error: Failed to execute 'insertBefore' on 'Node'

happytomatoe, 2024-09-17:
### Tool

None

### Interface

None

### What happened?

When I go to VM Emulator and press start I get 
[image]
Commenting [this line ](https://github.com/nand2tetris/web-ide/blob/b55db92c37881c23953a3081a03e948a87eba532/web/src/pages/vm.tsx#L411)helps. Is it only me experiencing this issue?

### Do you want to try to fix this bug?

- [ ] I want to try to add this feature!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

happytomatoe, 2024-09-17:
@netalondon Could you take a look when you have time?

DavidSouther, 2024-09-18:
@happytomatoe make sure you've run `npm run build` before you first run the web ide. If this continues, please use the Discussions board for local development issues. If this is happening after you've made changes to the code, do the same (`npm run build`) and then include a link to the branch you're working on in the discussion. Thanks!

closed 2024-09-18 by DavidSouther