nand2tetris/web-ide #481: [bug]: CPU-external.tst is checking outM during A instructions

jamesdarrenmuir, 2024-09-23:
### Tool

Hardware Simulator

### Interface

Website (https://nand2tetris.github.io/web-ide)

### Contact Details

jamesmuir@live.com

### What happened?

My CPU design passed `CPU.tst`, but failed `CPU-external.tst`, because the `outM` outputs didn't match during the A-instructions, where they are undefined anyways.

I tried the version of `CPU-external.tst` and the corresponding `CPU-external.cmp` files from the [Nand to Tetris Software page](https://www.nand2tetris.org/software) and they worked, because the `outM` values during A-instructions were wildcards.

I believe the fix is just to update the `CPU-external.tst` and `CPU-external.cmp` files to match the working ones I downloaded.

<details> 
<summary>CPU design:</summary>
<code>
CHIP CPU {

    IN  inM[16],         // M value input  (M = contents of RAM[A])
        instruction[16], // Instruction for execution
        reset;           // Signals whether to re-start the current
                         // program (reset==1) or continue executing
                         // the current program (reset==0).

    OUT outM[16],        // M value output
        writeM,          // Write to M? 
        addressM[15],    // Address in data memory (of M)
        pc[15];          // address of next instruction

    PARTS:
    // instruction decoder
    Mux16(a=false, b=instruction, sel=instruction[15],
    out[0]=jmpIfPos,
    out[1]=jmpIfZero,
    out[2]=jmpIfNeg,
    out[3]=writeM,
    out[4]=loadD,
    out[5]=loadA,
    out[6]=notOut,
    out[7]=doAdd, // 0 for and, 1 for add
    out[8]=notY,
    out[9]=zeroY,
    out[10]=notX,
    out[11]=zeroX,
    out[12]=useM, // 0 for A, 1 for M
    out[15]=cInstruction,
    );
    Not(in=cInstruction, out=aInstruction);

    // load A reg logic
    Or(a=aInstruction, b=loadA, out=ldA);

    // parts from Figure 5.8
    Mux16(a=instruction, b=aluOut, sel=cInstruction, out=aRegInputMuxOut); // A reg input MUX
    ARegister(in=aRegInputMuxOut, load=ldA, out=aRegOut, out[0..14]=addressM);
    Mux16(a=aRegOut, b=inM, sel=useM, out=yIn); // A vs M MUX

    DRegister(in=aluOut, load=loadD, out=xIn);

    ALU(x=xIn, y=yIn, zx=zeroX, nx=notX, zy=zeroY, ny=notY, f=doAdd, no=notOut, out=aluOut, out=outM, zr=zer, ng=neg);

    Not(in=jmp, out=notJmp);
    PC(in=aRegOut, load=jmp, inc=notJmp, reset=reset, out[0..14]=pc);

    // jump logic
    Or(a=zer, b=neg, out=notPos);
    Not(in=notPos, out=pos);
    And(a=jmpIfPos, b=pos, out=jmpPos);
    And(a=jmpIfZero, b=zer, out=jmpZr);
    And(a=jmpIfNeg, b=neg, out=jmpNg);
    Or8Way(in[0]=jmpPos, in[1]=jmpZr, in[2]=jmpNg, out=jmp);

}
</code>
</details>

### Do you want to try to fix this bug?

- [X] I want to try to add this feature!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jamesdarrenmuir, 2024-09-23:
I believe this issue is fixed by this PR: https://github.com/nand2tetris/web-ide/pull/433, but it seems that it hasn't been deployed to the live site yet.

DavidSouther, 2024-10-07:
That PR has been deployed, and I see the wildcards when I load the project in incognito mode. Try backing up your implementations with the download button, then, in Settings, click "Reset project files"?

closed 2024-11-30 by DavidSouther