nand2tetris/web-ide #487: [bug]: XOr behaves like Xor (should raise error)

eeeeeecsh, 2024-10-14:
### Tool

Hardware Simulator

### Interface

Website (https://nand2tetris.github.io/web-ide)

### What happened?

When I wrongly typed "XOr" where I should type "Xor", it can still run successfully (without raising any error) as the chip "XOr" work just like Xor.
But in the desktop version of the hardware simulator, the same code will get an error like "In HDL file XOr.hdl, Line 9, Chip name doesn't match the HDL name: load HalfAdder.hdl"
Below is an example of this bug.
[image]
I've also tried other similar typos like "XOR", only "XOr" behaves wrongly (others will raise an error as they should)

### Do you want to try to fix this bug?

- [ ] I want to try to add this feature!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DavidSouther, 2024-11-29:
Confirmed - in earlier versions of the web IDE, the chip was incorrectly named `XOr`. So if you loaded it both before and after that point, your IDE will have both `Xor` and `XOr` as builtin chips.

DavidSouther, 2024-11-29:
We'll add a step to remove the `XOr` chip when you next load the IDE.

DavidSouther, 2024-12-19:
Resolved in f95365f

closed 2025-02-12 by DavidSouther