nand2tetris/web-ide #488: [bug]: Err 14 when trying to run Output tests from for the last project provided files

happytomatoe, 2024-10-16:
### Tool

None

### Interface

None

### What happened?

How to recreate:
1) Run this 
```
// This file is part of www.nand2tetris.org
// and the book "The Elements of Computing Systems"
// by Nisan and Schocken, MIT Press.
// File name: projects/12/OutputTest/Main.jack

/** Test program for the OS Output class. */
class Main {

    /** Outputs the entire character set to the screen using all the
     *  methods of the Output class. */
    function void main() {
        var String s;
    
    	let s = String.new(1);
    	do s.appendChar(String.doubleQuote());
    
        do Output.moveCursor(0, 63);
        do Output.printChar(66);
        do Output.moveCursor(22, 0);
        do Output.printChar(67);
        do Output.moveCursor(22, 63);
        do Output.printChar(68);
        do Output.printChar(65);
        
        do Output.moveCursor(2, 0);
        do Output.printString("0123456789");
        do Output.println();
        
        do Output.printString("ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz");
        do Output.println();
        
        do Output.printString("!#$%&'()*+,-./:;<=>?@[\]^_`{|}~");
        do Output.printString(s);
        do Output.println();
        
        do Output.printInt(-12345);
        do Output.backSpace();
        do Output.printInt(6789);
        
        return;
    }
}

```
I get ERR14

### Do you want to try to fix this bug?

- [ ] I want to try to add this feature!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

happytomatoe, 2024-10-16:
If I remember correctly there is also another bug - If you run all provided files you can flush it out
https://github.com/nand2tetris/projects/tree/main/12

HofmT, 2024-12-25:
I also have problems to run Output tests however in my case it returns `Runtime error: Maximum call stack size exceeded`.
I tried to reproduce this bug by running the web-ide locally on my mac but somehow that works fine. I tested latest main branch with commit ID `cd2963f4ad14c1f8ce99ae0f11d5a749cc28bc0c`

DavidSouther, 2026-10-22:
Thank you for the detailed report, @happytomatoe, and for the follow-up from @HofmT.

We could not reproduce this on current `main`. The program from your report (the `projects/12/OutputTest` `Main.jack`) now compiles and runs to completion with exit code 0, using the built-in OS. The regression test is here, in #695:

https://github.com/nand2tetris/web-ide/blob/564f9fc/simulator/src/repro/issue-488.test.ts

Both reports predate our move to Vite, and both note that the program behaved differently on the deployed site than locally (@HofmT could not reproduce it on a local checkout either). That points at a stale or cached build on the site rather than at the simulator. We are tracking the service-worker side of that separately in #694.

I'm closing this as can't reproduce. If you still see ERR 14 or a "Maximum call stack size exceeded" error on the current site, please open a new issue with your browser and version and the exact files you ran, and we'll take another look.

---
_Generated by [Claude Code](https://claude.ai/code)_

closed 2026-10-25 by DavidSouther