nand2tetris/web-ide #490: [bug]: Web IDE couldn't load '.hack' files in test scripts of projects/5/Computer.hdl

abhishekrathore, 2024-10-17:
### Tool

Hardware Simulator

### Interface

Website (https://nand2tetris.github.io/web-ide)

### Contact Details

rathore.abhishek87@gmail.com

### What happened?

Facing a bug when running the "projects/5/Computer.hdl" tests.
In the chrome console, it shows that 'Add.hack' or 'Max.hack' file is not loaded due to some exception.

[image]

Digging deeper in source code found  that code is searching for file in '/samples/' directory NOT '/projects/5' directory.
Fixed it in the Chrome local storage and then it worked.

[image]

Its only for files with ".hack" extensions. I guess the '/samples' directory key in WEB IDE is not there by default.

### Do you want to try to fix this bug?

- [ ] I want to try to add this feature!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

fwoerister, 2024-10-17:
+1

I encountered the same error. Would be interesting if the hack files are loaded to the wrong directory or the ide is looking at the wrong location when loading hack files to the ROM.

lyrakisk, 2024-11-07:
I also had the same problem. The quick fix that @abhishekrathore suggested solved it locally for me as well.

EricOrrDev, 2024-11-18:
I am having same problem, working on a resolution for firefox users.

DavidSouther, 2024-11-30:
This should be resolved now that #450  is merged and deployed.

closed 2024-11-30 by DavidSouther