nand2tetris/web-ide #502: [bug]: Test scrip stuck on loading binary program Add.hack on Hardware Simulator

busoff, 2024-11-21:
### Tool

Hardware Simulator

### Interface

Website (https://nand2tetris.github.io/web-ide)

### Contact Details

nobusoff@gmail.com

### What happened?

When I try to run Built-in Computer chip (Chapter 5) on hardware simulator , the test script stuck on "Loads the binary program Add.hack into the computer's instruction memory" and can't continue.

I tried Chrome Version 131.0.6778.70 and Edge Version 131.0.2903.51 and the both failed. What did I miss?
[image]

### Do you want to try to fix this bug?

- [ ] I want to try to add this feature!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

filakhtov, 2024-11-21:
I just literally ran into this exact issue. Looking into dev console in Chrome reveals: 
[image]

filakhtov, 2024-11-21:
Okay, debugging even further, I noticed that my local storage did not have the `.hack` files. I made a backup of my HDL's, cleared the storage and refreshed the page – files are now there and the emulator works as expected.

filakhtov, 2024-11-21:
Okay, made it work by using a local override for:
https://github.com/nand2tetris/web-ide/blob/634686eca2bbb6c0f795bf72f1e2f440647f02e5/simulator/src/test/instruction.ts#L222
and changing the directory to `test.fs.pushd("/projects/05");`, hopefully this would help to track the issue down.

DavidSouther, 2024-11-30:
This should be resolved with the latest deployment after #450 was merged; please re-open if it persists.

closed 2024-11-30 by DavidSouther