nand2tetris/web-ide #526: [bug]: Stack overflow with simple code

doublethink13, 2024-12-05:
### Tool

Jack Compiler

### Interface

Website (https://nand2tetris.github.io/web-ide)

### What happened?

I'm compiling and running this code:

```
class Main {
    function void main() {
        while (true) {
            do Output.printString("Hello, world!");
            do Output.println();
            do Sys.wait(10000);
            do Screen.clearScreen();

            do Output.printString("Goodbye, world..");
            do Output.println();
            do Sys.wait(10000);
            do Screen.clearScreen();
        }

        return;
    }
}
```

I have two issues with it.

First, the Sys.wait(10000) instruction does not wait the expected time.

Secondly, the code eventually reaches a stack overflow.

See:

[image]

### Do you want to try to fix this bug?

- [ ] I want to try to add this feature!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

doublethink13, 2024-12-07:
Note that the following code, does not produce a stack overflow for me:

```
class Main {
    function void main() {
        var int test;

        while (true) {
            let test = test + 1;
            do Sys.wait(1000);
        }

        return;
    }
}
```

The only local variable just keeps on increasing. However, the `do Sys.wait(1000)` still does not work.

doublethink13, 2024-12-07:
I'll test this in the older local compiler to see if the problem persists once I have the time.

dorianignee, 2024-12-10:
The message "Maximum call stack size exceeded" is shown when the text cursor is past the end of the screen. I don't really know why the error occurs, because the "Output" class should just reset the row to zero, when it does a newline in the last row.

dorianignee, 2024-12-10:
I created a PR for the error message. The Sys.wait instruction will be fixed with PR #528. 
When those two issues are fixed, there's still a problem with your code: You create new String objects with every printString command. So you will run out of heap space (ERR6) after a few thousand iterations. You could reuse your strings to avoid that problem like this:

`class Main {
    function void main() {
        var String hello, goodbye;
        let hello = "Hello, world!";
        let goodbye = "Goodbye, world..";

        while (true) {
            do Output.printString(hello);
            do Output.println();
            do Sys.wait(10000);
            do Screen.clearScreen();

            do Output.printString(goodbye);
            do Output.println();
            do Sys.wait(10000);
            do Screen.clearScreen();
        }

        return;
    }
}`

EduardoMunozSepio, 2024-12-10:
Unfortunately, that still doesn't work:

[image]

doublethink13, 2024-12-10:
Unfortunately that still does not work:

[image]

dorianignee, 2024-12-10:
Yes, you need to wait until my PRs #529 and #528 are merged into this repo. I'm just a random guy that contributes to this project, but I can't update this repo directly. One of the maintainers needs to review and approve my changes and merge my pull requests, so that they become active on the website. 
As long as this issue is open, my changes are not applied to the page.

doublethink13, 2024-12-10:
I appreciate your help! Thanks a lot :)

closed 2024-12-19 by DavidSouther

doublethink13, 2025-01-02:
I see that you released a new version [recently](https://github.com/nand2tetris/web-ide/releases/tag/2024.52.0) but I still get an old IDE version in my browser:

[image]

DavidSouther, 2025-01-02:
Could you paste the headers for the index.html request?

https://developer.chrome.com/blog/copy-the-response-of-a-network-resource-to-your-clipboard

doublethink13, 2025-01-03:
Now I'm getting the latest version:

[image]

Just for completion, here are the headers:

[image]

[image]

[image]

[image]

commit cee36e6a (2024-12-19): fix #526: vm emulator raises "call stack exceeded"… (#529)

fix #526: vm emulator raises "call stack exceeded" when doing println in last row

move check for last screen row from printChar to println, so the check runs on every newline command