nand2tetris/web-ide #527: [bug]: (Safari, CPU Emulator) Clearing the ROM after starting the execution of the test script.

pmokeev, 2024-12-08:
### Tool

CPU Emulator

### Interface

Website (https://nand2tetris.github.io/web-ide)

### What happened?

Sequence of steps:
1. Loading a local .hack file into the ROM (previously compiled in Assembler tool)
[image]
2. Loading a .tst script from the selected directory.
[image]
3. Verifying that everything is loaded successfully.
[image]
4. Clicking the Run button (or Step) in the test script resets the ROM block to its default state – everything is cleared.
[image]

Because of this, the test results show -1 everywhere.

This is likely due to Safari., because in Google Chrome everything is fine

### Additional Comments

Safari Version 17.6 (19618.3.11.11.5)

### Do you want to try to fix this bug?

- [ ] I want to try to add this feature!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hi-slava, 2025-09-10:
For me it currently doesn't work in Google Chrome as well as in Safari. 
Google Chrome Version 139.0.7258.155 (Official Build) (arm64)

I tried it all: 
- Loading a local file with the same name "Mult.asm"
- Loading the .hack file
- Sending the code from assembler tab by pressing "Load in CPU Emulator"

DavidSouther, 2026-10-21:
Thanks for the detailed steps. This matches #622, where Run clears the loaded ROM, and the Safari detail is noted there. Closing as a duplicate.

---
_Generated by [Claude Code](https://claude.ai/code)_

closed 2026-10-21 by DavidSouther