nand2tetris/web-ide #533: [feature]: Allow opening empty folders on Jack Compiler page

dorianignee, 2024-12-11:
### Tool

Jack Compiler

### Interface

Website (https://nand2tetris.github.io/web-ide)

### What feature are you proposing?

This feature was removed with commit 8484832af693b672d9fbc7ddfd1f4b62e135d8c7, but I liked the possibility to fiddle around with the Jack language by just creating an empty folder in the file chooser dialog, and then creating .jack files on the compiler page. 

Since I can't create an empty .jack file in the file chooser dialog, I have to open vscode with that folder, create an empty Main.jack file and then use the compiler page to write code. That's an annoying roundabout. 

Is there any particular reason to disallow empty folders on the compiler page, except for telling the user "Oops, you probably selected the wrong folder"?

On all the other pages, disallowing empty folders is reasonable, because they don't interact with the local file system directly. If you're ok with that, I'd like to remove the check for empty folders in a new PR.

### Do you want to try to add this feature?

- [X] I want to try to add this feature!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

DavidSouther, 2024-12-19:
I think this was an oversight more than an intentional reduction in behavior? Happy to review a PR that adds this back.

dorianignee, 2024-12-20:
I removed the check for empty folders in my local clone, but that didn't solve the issue. The issue was introduced with the "Release: FileSystem + UI" (8484832af693b672d9fbc7ddfd1f4b62e135d8c7). I've tested the project with the commit right before the release and it turns out, that it showed the same error message, but would still load the empty folder.

I guess, the problem is caused by the same bug, that also causes issue #524, but the commit is too large to figure out, what is causing the issue. Maybe I'll take another look at it after the christmas holidays.

DavidSouther, 2026-10-25:
If you come back to this, it looks like the underlying issue in #524 has been resolved!