nand2tetris/web-ide #542: [bug]: Testing StringTest with custom String.jack fails

HofmT, 2024-12-24:
### Tool

Jack Compiler

### Interface

Website (https://nand2tetris.github.io/web-ide)

### What happened?

I was testing my implementation of `String.jack` inside the folder `StringTest` which I uploaded to the Web IDE JackCompiler. The folder only consisted of the jack files: `String.jack` and `Main.jack`.

I noticed that whenever I printed a string, it would start with two whitespaces before printing any characters in the string., e.g. `abc` would become `  abc`.

I did some debugging and I think the error is related to the following:
Since my folder did not include the `Output.jack` file, it retrieves the code from the built-in files. When printing a string (`Output.printString()`) there is a dependency to the function `charAt` inside `String.jack`. I assume the compiler does not reference my provided `String.jack` but instead the `built-in` `String.jack` which has a different implementation of the String object.
As a result, the behavior of how the string is getting printed is not deterministic and results into arbitrary characters being printed.

I then tested my custom `String.jack` with my custom `Output.jack` in the same StringTest directory and then my code would behave correctly.
I'm wondering now if it is possible to test the String.jack in isolation without providing your custom `Output.jack` file.

### Do you want to try to fix this bug?

- [ ] I want to try to add this feature!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

elp2, 2025-01-14:
This also affects MemoryTest even worse since it calls Array.new, which if there's no Array.hack in MemoryTest, will fall back to the built-in memory.ts functions, completely ignoring the memory code you were trying to test!

My PR re-writes the string and os functions to use pointers and heap like a real hack program would.
To fix the MemoryTest, the project files could use Memory.alloc instead of Array.new.

A more cohesive fix would probably be to have compiled golden VMs for each OS function instead of bespoke JS functions (although it might be slower)?

rsiggy, 2025-03-26:
I am experiencing this exact same problem with the added characters to the start of a string when using my String class and the built-in Output class.  All works perfectly with my Output class, which has already passed the acceptance tests.  By adding additional field variables to my String class I can cause the built-in Output.printString output to have even more spaced added at the start of a line.  Of course, other class field variable should have zero effect on anything.  Hmm...   

I've found several String classes online that claim to work and pass the test and the code is functionally identical to mine.  I'm wondering if this is an issue with the web IDE and so may try the tests with the PC executable version of the VM simulator to see if it works there.  The String classes I found that claim to pass the tests are old enough that they were likely run on the old SW version, not the new web IDE version.

elp2, 2025-03-26:
Maybe it's not loading your classes are they in the right dir to get picked up? I only used the WebIDE to finish everything, you could look at my repo https://github.com/elp2/nand2tetris

rsiggy, 2025-03-26:
thanks, elp2 but I took a look at your repo and your String class has the exact same problem that HofmT and I are describing.  In your case, the output is only shifted one character to the right - that is presumably because you only have ONE other field variable besides your character Array that holds the actual string.  Looks like you are not storing the maximum length of the string and can therefore not do any bounds checking in your class.  Try adding another field variable to the class to hold maxLength and voila - your output is now shifted TWO characters to the right.  Every single "working" string class I've found on the web shows this exact same problem.

elp2, 2025-03-29:
It's been a while,  but I believe it's not the String class itself, it's how the fallback to the TypeScript versions of the other OS methods work. They manage the size/freelist through TS objects, rather than storing the (IIRC block size and next ptr). So that's what the two empty things are since it expects ASCII characters to be the first two bytes. If you patch my PR in it should be fixed since it expects those to be set like described in the docs. Or if you do enough of the other classes (Array and Memory in the same dir should be enough IIRC) to prevent the TS versions being used.

rsiggy, 2025-03-29:
Hi Edward.

 

Yes, I’ve absolutely convinced myself that this is a bug in the interactions between the built-in Output class and a user String class.   If I use both my String and Output classes all works fine – same is true for your classes in your repo.  Only time the problem presents itself is when running a user String and build-in Output.  In any case, if I use all of my own OS classes I pass all tests and Pong and other programs I’ve written work perfectly so I’m considering my course completed.  I would be nice to have the team address this failing of the simulator either by correcting the tools or via some sort of documentation so others won’t waste time trying to figure out what is going wrong with their String class, which is really a very simple and obvious bit of code.  Anyway, thanks for the interaction on this issue!

 

Robert

elp2, 2025-03-30:
That's what I decided too! 👍  Congrats on finishing!