nand2tetris/web-ide #543: [bug]: Cannot write to Mult.asm file

petrovj12, 2024-12-25:
### Tool

CPU Emulator

### Interface

Website (https://nand2tetris.github.io/web-ide)

### What happened?

I cannot write to the file Mult.asm, neither by editing it directly in the CPU Emulator nor by transferring from the Assembler. When the Mult.tst script tries to load the file Mult.asm it is loaded as only zero instructions.

### Do you want to try to fix this bug?

- [ ] I want to try to add this feature!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pbrw, 2024-12-26:
I faced the same issue. I wrote assembler code in the Web UI. Then I run a `Mult.tst` test but the line `load Mult.asm` just cleared my ROM and I can't find a way to get my solution back.

maueroats, 2024-12-26:
Clarification: in Project 4, the testing code begins with `load Mult.asm`, but when testing this code, the user will be uploading their `Mult.asm` file written somewhere else and either uploaded or copy-and-pasted into the left side code pane.  

Under these circumstances, the code that users want to test is not "in the file" Mult.asm. 

I see that the code *does* revert to some "saved" version.

Somehow, though, my "saved" version is something I myself wrote. I don't know what I did to make that happen. Perhaps the behavior changed recently? I see the ability to edit the test scripts seems to have disappeared, or there would be an easy fix - "comment out the load statement".

maueroats, 2024-12-26:
Please rename bug to be more informative: project 4 testing code re-loads asm files without including user changes

t1mp4, 2024-12-27:
> Clarification: in Project 4, the testing code begins with `load Mult.asm`, but when testing this code, the user will be uploading their `Mult.asm` file written somewhere else and either uploaded or copy-and-pasted into the left side code pane.
> 
> Under these circumstances, the code that users want to test is not "in the file" Mult.asm.
> 
> I observed that the assembly code _does_ revert to some "saved" version when the test script is run. In a "clean" browser, this file is empty / full of zeroes.
> 
> In one of my browsers, my "saved" version is something I myself wrote. I don't know what I did to make that happen. Perhaps the behavior changed recently?
> 
> I see the ability to edit the test scripts seems to have disappeared, or there would be an easy fix - "comment out the load statement".

Yeah, what's the reasoning behind disallowing the user to edit test scripts or upload their own test scripts?

t1mp4, 2024-12-27:
A temporary solution for now is to manually update localStorage in the browser console and comment out the "load" at the top.

```javascript
localStorage.setItem("/projects/04/Mult/Mult.tst", `// This file is part of www.nand2tetris.org
// and the book "The Elements of Computing Systems"
// by Nisan and Schocken, MIT Press.
// File name: projects/4/mult/Mult.tst

// Tests the Mult program, designed to compute R2 = R0 * R1.
// Tests the program by having it multiply several sets of
// R0 and R1 values.

//load Mult.asm,
compare-to Mult.cmp,
output-list RAM[0]%D2.6.2 RAM[1]%D2.6.2 RAM[2]%D2.6.2;

set RAM[0] 0,   // Sets R0 and R1 to some input values
set RAM[1] 0,
set RAM[2] -1;  // Ensures that the program initialized R2 to 0
repeat 20 {
  ticktock;
}
set RAM[0] 0,   // Restores R0 and R1 in case the program changed them
set RAM[1] 0,
output;

set PC 0,
set RAM[0] 1,   // Sets R0 and R1 to some input values
set RAM[1] 0,
set RAM[2] -1;  // Ensures that the program initialized R2 to 0
repeat 50 {
  ticktock;
}
set RAM[0] 1,   // Restores R0 and R1 in case the program changed them
set RAM[1] 0,
output;

set PC 0,
set RAM[0] 0,   // Sets R0 and R1 to some input values
set RAM[1] 2,
set RAM[2] -1;  // Ensures that the program initialized R2 to 0
repeat 80 {
  ticktock;
}
set RAM[0] 0,   // Restores R0 and R1 in case the program changed them
set RAM[1] 2,
output;

set PC 0,
set RAM[0] 3,   // Sets R0 and R1 to some input values
set RAM[1] 1,
set RAM[2] -1;  // Ensures that the program initialized R2 to 0
repeat 120 {
  ticktock;
}
set RAM[0] 3,   // Restores R0 and R1 in case the program changed them
set RAM[1] 1,
output;

set PC 0,
set RAM[0] 2,   // Sets R0 and R1 to some input values
set RAM[1] 4,
set RAM[2] -1;  // Ensures that the program initialized R2 to 0
repeat 150 {
  ticktock;
}
set RAM[0] 2,   // Restores R0 and R1 in case the program changed them
set RAM[1] 4,
output;

set PC 0,
set RAM[0] 6,   // Sets R0 and R1 to some input values
set RAM[1] 7,
set RAM[2] -1;  // Ensures that the program initialized R2 to 0
repeat 210 {
  ticktock;
}
set RAM[0] 6,   // Restores R0 and R1 in case the program changed them
set RAM[1] 7,
output;`);
```

Occccn, 2025-02-20:
Try translating the source into binary in the Assembler.
The translated binary can be executed in the CPU emulator.
https://nand2tetris.github.io/web-ide/asm

v3ai, 2025-04-01:
Had the same problem

commit 5d44c798 (2025-08-22): Allow users to upload local .tst and .cmp files in CPU Emulator (#591)

* Allow users to upload local .tst and .cmp files in CPU Emulator

Fixes #543
Fixes #565
Fixes #551
Fixes #543

---------
closed 2025-08-22 by DavidSouther